lufton/esphome-inv-8851 #20: Sensors missing on DTU WBS1-V001

JoseLu81, 2024-06-30:
Hello, first of all I wanted to thank you for this fantastic work. I have flashed the WiFi device and I already have it integrated with esphome in homeassistant. It works perfectly, but there are sensors that don't appear to me. 
PV power
PV current
PV voltage

I have tried updating the project but it still doesn't work.
The version you flash is the complete one, it is not the local one.

Do you have any idea what could happen to him?
Thank you so much.

lufton, 2024-06-30:
Hello, did you uncommented this line?
https://github.com/lufton/esphome-inv-8851/blob/a6329518e1a37ca1a8b233c5846a29ad54125db4/dtu-wbs1-v001-48v-example.yaml#L20C1-L20C27

JoseLu81, 2024-07-03:
Hello, I did realize that the file I was commenting on contained the sensors I was missing. Thank you very much for this fantastic project. Now I have all the sensors enabled. Can I ask one more question? The battery charging current is an entity that gives positive and negative numbers. An entity could be created for the charging current and another entity for the discharge current (and that was not a negative value). thanks for everything

lufton, 2024-07-03:
If there will be 2 sensors, then what should `battery_discharge_current` show during charging and vice versa? 0? Negative value? This is counterintuitive. Now sensor named `battery_charge_current` not `battery_current` witch makes clear that positive value means charging and negative discharging.

JoseLu81, 2024-07-25:
Te importaría decirme cómo hacer la.plantilla en Homeassistant?? He estado buscando información y no logro entender cómo hacerlo.
Muchas gracias

lufton, 2024-07-25:
@JoseLu81, Forest of all go to Settings / Devices and services / Helpers. There add a template sensor. Give it a name (e.g. `Battery discharge current`) and provide template in Jinja2 style. In your case that should be something like:
```
{% set value = states('sensor.dtu_wbs1_v001_battery_charge_current') | float %}
{{ -value if value < 0 else None }}
```
The same way you can do with charge current:
```
{% set value = states('sensor.dtu_wbs1_v001_battery_charge_current') | float %}
{{ value if value > 0 else None }}
```
In this way sensor will only have value if it makes sense (positive for charge current and negative for discharge current).

JoseLu81, 2024-07-27:
Thank you very much, I have tried it and it works perfectly. Thank you for everything, I appreciate the time you have dedicated to me and especially the time developing the project. A hug from Spain (Valencia).

lufton, 2024-07-27:
¡Arriba España!
Glory to Ukraine!

closed 2024-08-28 by lufton