lufton/esphome-inv-8851 #21: Error - Must contain at least one of tx_pin, rx_pin.

balistic2013, 2024-07-03:
Hello, first of all I wanted to thank you for this fantastic work. I have flashed the WiFi device with ESPHome. 
But I am unable to flash my .yaml below. I get the folloiwng error:

"Failed config
uart: [source /data/packages/d445f241/packages/core.yaml:15]
Must contain at least one of tx_pin, rx_pin.
- baud_rate: 9600"

"DTU.yaml"

substitutions:
  inverter_maximum_power: "4500"
  name: dtu-wbs1-v001
  friendly_name: DTU WBS1-V001

esp32:
  board: esp32-c3-devkitm-1

uart:
  baud_rate: 9600

inv_8851:

sensor:
  - platform: inv_8851
    battery_voltage:
      name: "Battery voltage"

packages:
  inv_8851:
    url: https://github.com/lufton/esphome-inv-8851
    ref: main
    files:
      - packages/core.yaml
      - packages/inverter.yaml
      - packages/grid.yaml
      - packages/temperature.yaml
      - packages/battery.yaml
      - packages/battery-24v.yaml
      # - packages/pv.yaml
      # - packages/parallel.yaml
      # - packages/bms-24v.yaml
      - packages/dtu_wbs1_v001.yaml
    refresh: 0s
      
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ap:
  ssid: ${friendly_name}
  password: !secret ap_password

api:
  encryption:
    key: !secret api_encryption_key

ota:
  - platform: esphome
    password: !secret ota_password

If you could please help me fix it. I'm sure it is something simple that I am missing.

balistic2013, 2024-07-03:
Hi 

Thanks for your reply. I have edited my post above hopefully it is easier to read. 
I am trying to use a copy of dtu-wbs1-v001-24v-example.yaml.
I will lookup how to clear the .esphome folder.

Thanks

lufton, 2024-08-28:
Is your issue resolved?

balistic2013, 2024-12-18:
Hi,

Sorry for the late reply. Yes my issue was resolved. Thanks

But I now have a new issue, when trying to Validate my .yaml, I am now getting this error.

INFO ESPHome 2024.12.0
INFO Reading configuration /config/esphome/test-device.yaml...
INFO Updating https://github.com/lufton/esphome-inv-8851@main
WARNING Found '${version}' (see inv_8851->version) which looks like a substitution, but 'version' was not declared
WARNING Found '${version}' (see inv_8851->version) which looks like a substitution, but 'version' was not declared
INFO Updating https://github.com/lufton/esphome-inv-8851.git@main
Failed config

inv_8851: [source /data/packages/d445f241/packages/core.yaml:26]
  
  expected int.
  version: ${version}
  update_interval: 5s

Any help much appreciated.