lufton/esphome-inv-8851 #22: Invalid header

Th3M1k3y, 2024-07-21:
Maybe a new version of the dongle released?

After flashing mine with esphome it only shows
```
[19:26:18]invalid header: 0xffffffff
[19:26:18]invalid header: 0xffffffff
[19:26:18]invalid headeESP-ROM:esp32c3-api1-20210207
[19:26:18]Build:Feb  7 2021
[19:26:18]rst:0x7 (TG0WDT_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
[19:26:18]Saved PC:0x4004d1f8
[19:26:18]invalid header: 0xffffffff
[19:26:18]invalid header: 0xffffffff
[19:26:18]invalid header: 0xffffffff
[19:26:18]invalid header: 0xffffffff
[19:26:18]invalid header: 0xffffffff
[19:26:18]invalid header: 0xffffffff
[19:26:18]invalid header: 0xffffffff
[19:26:18]invalid header: 0xffffffff
[19:26:18]invalid header: 0xffffffff
[19:26:18]invalid header: 0xffffffff
```

This is the chip on mine
```
esptool.exe -p COM3 read_mac
esptool.py v3.2
Serial port COM3
Connecting.......................
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
```
Text on it is `ESP32-C3-MINI-1`

Th3M1k3y, 2024-07-21:
I made some progress. I had to use this to get it to successfully get it to run ESPHome

```
substitutions:
  inverter_maximum_power: "4500"
  name: solar-inverter
  friendly_name: Solar Inverter
  uart_tx_pin: GPIO4
  uart_rx_pin: GPIO5

esphome:
  platformio_options:
    board_build.f_flash: 40000000L
    board_build.flash_mode: dio
    board_build.flash_size: 4MB

esp32:
  variant: ESP32C3
  board: esp32-c3-devkitm-1

packages:
  inv_8851:
    url: https://github.com/lufton/esphome-inv-8851
    ref: main
    files:
      - packages/core.yaml
      - packages/inverter.yaml
      - packages/grid.yaml
      - packages/temperature.yaml
      - packages/battery.yaml
      - packages/battery-24v.yaml
      # - packages/pv.yaml
      # - packages/parallel.yaml
      # - packages/bms-24v.yaml
    refresh: 0s

wifi:
  ssid: !secret iot_ssid
  password: !secret iot_pass

  ap:
    ssid: '${friendly_name}'
    password: !secret fallback_password

api:
  encryption:
    key: !secret api_encryption

ota:
- platform: esphome
  password: !secret esphome_ota_password
```

Pin numbers has changed too. So far I found the TX and RX pin, will see if I can find the ones for the LEDs tomorrow.

Th3M1k3y, 2024-07-21:
LEDs got fixed by including `- packages/dtu_wbs1_v001.yaml`

closed 2024-07-21 by Th3M1k3y

lufton, 2024-07-21:
Great! Glad you was successful!