lufton/esphome-inv-8851 #23: Installation....

korishan, 2024-08-18:
I'm new to the whole Esphome scene. I have managed to get Emporia moved over to it and works pretty good. I have a PowMr 5k Hybrid inverter. I've already installed the esphome and did the wiring to hook up the unit. But the Esp reports "nan".

It was recommended to try out your software to see if the model was unsupported by odya/esphome-powmr-hybrid-inverter

But for the life of me, I can not figure out how to get your code working.  Do I have to reflash the esp with a new copy of esphome?? i was thinking not as that's just a base (I think). 
Every time I try to do a compile, I get different kinds of errors. I've tried moving directories, explicitly declaring paths, changing file locations. Nothing worked.

I'm very confused on this after working on it for several hours over 2 days. Appreciate any help to get this working.

lufton, 2024-08-18:
Current project is only compatible with 4500 and 6500 models. Those, that works with DTU like [this](
[image]
). If yours is different, then most likely you need to use odya/esphome-powmr-hybrid-inverter project.

cristianoggi, 2024-08-18:
Salve. Ho un inverter Y&H con un dispositivo di rete modello DTU-WBS1-V001. Come posso integrarlo in Home Assistant? Grazie.

korishan, 2024-08-18:
> Salve. Ho un inverter Y&H con un dispositivo di rete modello DTU-WBS1-V001. Come posso integrarlo in Home Assistant? Grazie.

Please start your own thread and not hijack someone elses. Thanks

> Current project is only compatible with 4500 and 6500 models. Those, that works with DTU like this. If yours is different, then most likely you need to use odya/esphome-powmr-hybrid-inverter project.

It was suggested I come here and try yours to see if it would work because of the messages I was getting with odya's code. I don't have a DTU device. I see your code can be used with the esp32.
I have no way of knowing if these are compatible or not, I figured I'd at least try and see what the output/readings are.

cristianoggi, 2024-08-19:
My inverter is a 4500. Is it possible to have an example in YAML? What integration should I download?

lufton, 2024-08-19:
@korishan if you don't have a DTU device you can make your own or buy one online. Again, let me reiterate: this project is for 4500W and 6500W models with glass front panel, I don't know what is your model and if this protocol is compatible with other models. Buying original DTU and flashing it is preferable as it is easier. But you can make your own according to odya's project, he even have [custom PCBs](https://github.com/odya/esphome-powmr-hybrid-inverter/tree/main/pcb) for that. The connection is basically the same, you need to connect corresponding pins of RJ45 to MAX3232 board and outputs of MAX3232 to ESP32/ESP8266. Then adjust example config (if select different RX and TX pins) and flash it. Also I think you would need to power up your ESP separately as MAX3232 requires additional 3.3V.

lufton, 2024-08-19:
@cristianoggi if you already have DTU-WBS1-V001 unit you can flash it using instructions on the main page of this project and [example file](https://github.com/lufton/esphome-inv-8851/blob/main/dtu-wbs1-v001-24v-example.yaml). All you need is to solder couple pins, connect USB-TTL module, enter device in boot mode and flash it using command I provided in README.

cristianoggi, 2024-08-19:
@lufton io l'unità DTU-WBS1-V001 ce l'ho. Il modulo USB/TTL lo devo collegare ad un pc scaricare il comando e per flasharlo? 

Nei valori "secret" ci inserisco i relativi valori?
 wifi:
  ssid: !**secret** wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: ${friendly_name}
    password: !**secret** ap_password

api:
  encryption:
    key: !**secret** api_encryption_key

ota:
  password: !**secret** ota_password

cristianoggi, 2024-08-19:
@lufton che integrazione devo scaricare per far funzione tutto?

lufton, 2024-08-19:
@cristianoggi yes you should use commands I described on the main page of this repo. Regarding secrets: you can either replace `!secret ap_password` with real password, but better to put it in `secrets.yaml` file next to configuration file.

korishan, 2024-08-20:
> I don't know what is your model and if this protocol is compatible with other models

I'm using the POW-SunSmart SP5K. 
I'm starting to wonder if the pinout is correct to connect the RJ45 cable to the MAX3232 with this model of inverter. The chip is getting pretty toasty after about 10mins. I did a check of the pins and Pin1/2 is registering 29.2mV
Using Brown (Pin8) as Reference I get:
Pin 1: 0mV
Pin 2: 29.2mV
Pin 3: -2.3V
Pin 4: -2.3V
Pin 5: -2.3V
Pin 6: -2.3V
Pin 7: 29.2mV

And if I use any other wire for Ref to any other wire (other than P8) I get jumping mVoltages. 
But thing is, this should be the same pinout as they use the same DTU adapter 🤔

lufton, 2024-08-20:
Actually I see that DTU you should use with this inverter is [LSW-5](https://powmr.com/products/powmr-wifi-module-wifi-hf-n?pr_prod_strat=pinned&pr_rec_id=5340990a7&pr_rec_pid=9268264993063&pr_ref_pid=8319037866279&pr_seq=uniform). It is completely different DTU as I see.
Check what I found https://github.com/StephanJoubert/home_assistant_solarman/issues/337

korishan, 2024-08-20:
Oooooo   Look what I found in the manual:
[image]

So I guessed correctly, the pins are different!!!  Time to do some testing and change the wiring pinouts!

cristianoggi, 2024-08-20:
@lufton Scusami ma i file dtu-wbs1-v001-24v-example.yaml li devo salvale nella cartella "config"? L'integrazione la devo aprire in ESPHome? Devo per forza flashare il dispositivo DTU? 

[image]

Grazie

korishan, 2024-08-20:
I'll just create a new thread with my findings

closed 2024-08-20 by korishan

lufton, 2024-08-21:
@cristianoggi you need:
1. Download example file
2. Put it in esphome configs folder (depend on type of installation) or create new device via UI and then just replace contents of a file to text from example file
3. Make adjustments if needed
4. Make changes in secrets.yaml file or put your credentials directly in config file
5. Connect TTL-USB module to DTU pins according to instructions in README
6. Put DTU to flashing mode (boot)
7. Flash config to DTU

cristianoggi, 2024-08-21:
@lufton Innanzitutto grazie per tutte le informazioni.
[image]

Ok?

lufton, 2024-08-22:
If you are about 3 commented lines, then yes. You can uncomment them if needed. My inverter (as probably many other cheap ones) does not support BMS communication and parallel connection, also not everyone have a PV field, that’s why lines are commented out.

cristianoggi, 2024-08-22:
@lufton  Ok. Li inserisco sempre nella cartella config/esphome?

lufton, 2024-08-22:
@cristianoggi if you are talking in context of Home Assistant, then yes, `config/esphome` is a valid folder for ESPHome config files. You should put one in that folder and you should be able to see this config in ESPHome UI.

cristianoggi, 2024-08-22:
Si sto parlando nel contesto Home Assistant. In questa cartella ho messo il file esempio di dtu-wbs1-v001-24v-example.yaml.
Appena posso devo fare la procedura per flashare il dispositivo.

lufton, 2024-08-22:
Correct.

cristianoggi, 2024-08-23:
@lufton  Come faccio a Mettere DTU in modalità flashing (avvio)? Il VCC lo iimposto +3.3V? Una volta scaricato il ponticello  BOOT1 collegato a GND1 lo stacco? Grazie.

lufton, 2024-08-23:
Connect you TTL module to DTU
| USB TTL              	        | DTU 	|
|----------------------	|-----	|
| 5V (next to RJ45)    	| 5V  	|
| TXD1 (next to ESP32) 	| RX  	|
| RXD1 (next to ESP32) 	| TX  	|
| GND1 (next to ESP32) 	| GND 	|

Also keep BOOT and GND connected until flash process is completed.

cristianoggi, 2024-08-23:
Il DTU quando viene collegato ak USB-TTL si accende il led power? Esptool.exe non si apre

lufton, 2024-08-23:
If it is in a boot mode, then no. It only blinks when default firmware is running in a normal loop.

lufton, 2024-08-24:
Did you download the tool? You need to download and extract archive. Then run the command from extracted folder or add esptool folder to system environment PATH variable.

cristianoggi, 2024-08-25:
@lufton Scusami. Lo strumento è stato scaricato ed estratto. Quando eseguo il comando dalla cartella estratta non parte, mi dice non eseguibile. Come posso aggiungere la cartella alla variabile PATH? Grazie.

lufton, 2024-08-25:
When you install esphome based firmware for the first time you need to do it via TTL COM port. When you run the command without `—device` parameter it should ask you how you want to install firmware. Please select COM port of your TTL module.

cristianoggi, 2024-08-25:
@lufton  Installato tutto. Un ultimo problema: le entità non le riconosce...

cristianoggi, 2024-08-25:
I installed everything. The device is ONLINE. One last problem: it doesn't recognize the entities... I even added them in YAML...

lufton, 2024-08-26:
This is weird, I think you could try debugging according to thread [here](https://github.com/lufton/esphome-inv-8851/issues/19). It has similar issue.