lufton/esphome-inv-8851 #28: Esp32 Ethernet eth-1

JoseLu81, 2024-09-08:
Hello friend, I am having communication problems with the original wifi dongle flashed with your firmware. My wifi doesn't work well and you disconnect. I have thought about putting in an esp32 ethernet. would it be possible? I mean removing the original wifi dongle and putting in an esp32 ethernet. Greetings and a thousand thanks.

lufton, 2024-09-08:
Does the problem appears only with custom firmware or stock had the same issue?
You can use Ethernet version, but then you also will need additional RS232 module (e.g. max3232) connected between ESP32 and inverter.

lufton, 2024-09-08:
Although connection should look like this: https://github.com/odya/esphome-powmr-hybrid-inverter?tab=readme-ov-file#connection

JoseLu81, 2024-09-13:
Good afternoon I was looking for more information about the hardware I needed to pass data over Ethernet. The wifi dongle fails because of wifi coverage, not because of the firmware. For now I am going to improve the wifi coverage. Thanks for the information

JoseLu81, 2024-09-13:
I would like to tell you something that happened to me in the inverter and it almost certainly has nothing to do with this fantastic project. Can we talk on Telegram? I don't speak English well at all, but I can use the translator. thank you

lufton, 2024-09-13:
Sure, ping me https://t.me/lufton

closed 2024-09-13 by lufton