lufton/esphome-inv-8851 #29: ESPHome unknown etities

PZotya, 2024-10-19:
Hi!

Buiding az ESPHome all entities of my device has unknown state.
[image]

My hardware setup:
The SBC is an ESP 8266 Wemos D1 mini, the power source is an 5V buck converter feeded form the inverter's pin 4 and 8.
The data cables are connected to the Wemos RX and DX pins, through a max232 board, which has a a power source the 3.3V of D1 mini.
The esphome version is 2024.9.2, Firmware: 2024.9.2 (Oct 19 2024, 12:05:21)

The inverter is a POW-HVM6.5K-48V.

PZotya, 2024-10-19:
This is my yaml config:
substitutions:
  inverter_maximum_power: "6500"
  version: "2"
  name: powmr
  friendly_name: "PowMr Inverter"
  uart_tx_pin: GPIO1
  uart_rx_pin: GPIO3

esp8266:
  board: d1_mini

packages:
  core: !include packages/core-local.yaml
  inverter: !include packages/inverter.yaml
  grid: !include packages/grid.yaml
  temperature: !include packages/temperature.yaml
  battery: !include packages/battery.yaml
  battery-48v: !include packages/battery-48v.yaml
  # pv: !include packages/pv.yaml
  # parallel: !include packages/parallel.yaml
  # bms-48v: !include packages/bms-48v.yaml

wifi:
  ssid: "yyyyyyyy"
  password: "xxxxxx"

  ap:
    ssid: "Powmr Fallback Hotspot"
    password: "RDmLwgXZMgYm"

api:

ota:
  - platform: esphome
    password: "xxxxxxxx"
logger:
  baud_rate: 0

lufton, 2024-10-22:
Hello @PZotya
- Make sure TX and RX of max232 are connected crisscross to RX and TX of ESP8266
- Make sure corresponding data lines are connected to corresponding max232 inputs
- Try change version parameter to "1" in `substitutions` section
- Try [enable logging](https://github.com/lufton/esphome-inv-8851?tab=readme-ov-file#debugging) and check if ESP8266 receives any data from inverter

I didn't test setup like that with external max232 controller, there may be some pitfalls with such hardware setup. But as far as I investigated original DTU is using similar setup, so in theory everything should work.

PZotya, 2024-10-22:
Thanks, changing the max232 datalines at ESP side solved the problem. Funny, but the working setup is that the RX port of ESP has to be connected to the max232 with arrow from ESP, and TX has to be connected to the max232 arrow to ESP.

lufton, 2024-10-23:
Great. Glad you solved the issue. Closing.

closed 2024-10-23 by lufton