lufton/esphome-inv-8851 #32: UART Protocol Mismatch (Received 00 00 Instead of 88 51)

adinetech, 2025-01-05:
Hello, I used GPIO1 (TXD) and GPIO3 (RXD) as per the requirement. TXD is connected to the RXD and vice versa but It doesn't seem to work. So I have changed this way,

```
MAX232 (TXD) --> GPIO3 (RXD) changed to GPIO5 (D1)
MAX232 (RXD) --> GPIO1 (TXD) changed to GPIO4 (D3)
```

UART
```
uart_tx_pin: GPIO4
uart_rx_pin: GPIO5
```
Pinout Reference.
[image]

It doesn't seem to return anything, so I swapped the tx and rn pins this way
```
uart_tx_pin: GPIO5
uart_rx_pin: GPIO4
```
I'm receiving the wrong protocol error (00 00) instead of the expected 88 51.
[image]

Any help is appreciated! The display of my Powmr 4500w 24v model is broken, so I can't change any of it's settings and it's charging my Lifepo4 pack to 28.7v.

lufton, 2025-01-06:
Hello, @adinetech, did you try https://github.com/lufton/esphome-inv-8851?tab=readme-ov-file#debugging?

adinetech, 2025-01-06:
Hey, thanks for the response! I haven’t done the troubleshooting yet. Unfortunately, my MAX3232 was damaged during soldering, and since I’ll be out of town starting tomorrow, I couldn’t perform any testing. For now, I had to let the cells charge at 28.7V (which I can’t prevent). I’ll revisit this after a few months if further help is needed.

I actually have this as well ([Amazon](https://www.amazon.in/dp/B0CD5STJHQ)). I cut the RJ45 wire to match the inverter's TXD and RXD, but it doesn’t seem to work. Do you have any ideas on this?

lufton, 2025-01-06:
So I believe you should be able to open serial monitor for COM port this creates. If you connect correctly you should be able to run command `88510003000000004d08` to retrieve state response and `88510003020000004cb0` to retrieve config response.
Responses should start with 8851.
You can read more about this protocol here: https://github.com/leodesigner/powmr4500_comm/tree/main

adinetech, 2025-01-06:
Thank you! I’ll reopen this if I still need help with the ESP setup,

closed 2025-01-06 by adinetech